kbeckmann/game-and-watch-retro-go #2: Cloning repository failed

ntrpnr, 2020-11-30:
I ran `git clone --recurse-submodules https://github.com/kbeckmann/game-and-watch-retro-go`

```
Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:kbeckmann/retro-go-stm32.git' into submodule path '/home/erik/game-and-watch-retro-go/retro-go-stm32' failed
Failed to clone 'retro-go-stm32'. Retry scheduled
Cloning into '/home/erik/game-and-watch-retro-go/retro-go-stm32'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:kbeckmann/retro-go-stm32.git' into submodule path '/home/erik/game-and-watch-retro-go/retro-go-stm32' failed
Failed to clone 'retro-go-stm32' a second time, aborting
```

kbeckmann, 2020-11-30:
Please try cloning again. Same issue as #1, which I have fixed now.

closed 2020-11-30 by kbeckmann

ntrpnr, 2020-11-30:
Worked. 👍

commit e65e0530 (2020-11-30): Change submodule to use https

Fixes #1, Fixes #2