kbeckmann/game-and-watch-retro-go #4: error during make

KsAmJ, 2020-12-01:
used stm32cubemx and followed and build the flasher 1st with success but when make on this getting errors below

`arm-none-eabi-gcc -x assembler-with-cpp -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32H7B0xx -ICore/Inc -IDrivers/STM32H7xx_HAL_Driver/Inc -IDrivers/STM32H7xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32H7xx/Include -IDrivers/CMSIS/Include -IDrivers/CMSIS/Include  -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/startup_stm32h7b0xx.d" startup_stm32h7b0xx.s -o build/startup_stm32h7b0xx.o
arm-none-eabi-gcc build/main.o build/stm32h7xx_it.o build/stm32h7xx_hal_msp.o build/stm32h7xx_hal_cortex.o build/stm32h7xx_hal_ltdc.o build/stm32h7xx_hal_ltdc_ex.o build/stm32h7xx_hal_rcc.o build/stm32h7xx_hal_rcc_ex.o build/stm32h7xx_hal_flash.o build/stm32h7xx_hal_flash_ex.o build/stm32h7xx_hal_gpio.o build/stm32h7xx_hal_hsem.o build/stm32h7xx_hal_dma.o build/stm32h7xx_hal_dma_ex.o build/stm32h7xx_hal_mdma.o build/stm32h7xx_hal_pwr.o build/stm32h7xx_hal_pwr_ex.o build/stm32h7xx_hal.o build/stm32h7xx_hal_i2c.o build/stm32h7xx_hal_i2c_ex.o build/stm32h7xx_hal_exti.o build/stm32h7xx_hal_ospi.o build/stm32h7xx_hal_sai.o build/stm32h7xx_hal_sai_ex.o build/stm32h7xx_hal_spi.o build/stm32h7xx_hal_spi_ex.o build/stm32h7xx_hal_tim.o build/stm32h7xx_hal_tim_ex.o build/system_stm32h7xx.o build/startup_stm32h7b0xx.o -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -specs=nano.specs -TSTM32H7B0VBTx_FLASH.ld  -lc -lm -lnosys  -Wl,-Map=build/gw_retrogo.map,--cref -Wl,--gc-sections -o build/gw_retrogo.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build/main.o: in function `Error_Handler':
/home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:632: undefined reference to `lcd_backlight_off'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:634: undefined reference to `lcd_backlight_on'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: build/main.o: in function `main':
/home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:148: undefined reference to `lcd_init'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:168: undefined reference to `OSPI_Init'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:169: undefined reference to `OSPI_EnableMemoryMappedMode'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:191: undefined reference to `app_main'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:187: undefined reference to `framebuffer1'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:187: undefined reference to `_sram_text'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:187: undefined reference to `_eram_data'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /home/ksamj/Downloads/game-and-watch-retro-go/Core/Src/main.c:187: undefined reference to `_siramdata'
collect2: error: ld returned 1 exit status
make: *** [Makefile:178: build/gw_retrogo.elf] Error 1
`
what would i missed?

KsAmJ, 2020-12-01:
my bad the issue that i didn't build the nes game with the make command

but still i can't manage to flash as i am getting this error 

`make -f Makefile.nes flash_extmem stlink
../game-and-watch-flashloader/flash.sh build/gw_retrogo_nes_extflash.bin
Loading image into RAM...
Open On-Chip Debugger 0.10.0+dev-snapshot (2020-08-19-11:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: No J-Link device found.

make: *** [Makefile.nes:293: flash_extmem] Error 1
`

i am using a stlink not J-link so have to pass that or change the command behavior to stlink?

ntrpnr, 2020-12-01:
Modify **adapter_config.cfg** in **game-and-watch-flashloader** directory to look like this:

```
source [find interface/stlink.cfg]
transport select hla_swd
source [find target/stm32h7x.cfg]
init
#reset
#halt
```

According to @kbeckmann you should also modify **flash.sh** in **game-and-watch-flashloader**:
https://twitter.com/kbeckmann/status/1333560671041155073

> If that's flashloader, then you probably might want to change http://flash.sh and edit the line 
> `    -c "resume; exit;"`
> into
> `    -c "resume; resume; exit;"`
> 
> Just found this issue which broke my brain today. Seems to be an st-link or openocd bug/behavior. Idk.

closed 2020-12-05 by kbeckmann