kbeckmann/game-and-watch-retro-go #6: Power button is not working

joraco-dev, 2020-12-05:
Seems that my power button stop working since emulator is flashed.

Any suggestions?

Thanks

nezza, 2020-12-05:
Sleep should be implemented in HEAD, which commit did you build?

kbeckmann, 2020-12-05:
Please join the discord and ask in the support channel as more people can see your issue and help you out. https://discord.gg/4rnjTeGF

Can you clarify exactly what is happening?

Can you attach the debugger?

Is the screen turning on at all?

Have you tried power cycling the debugger and target?

joraco-dev, 2020-12-05:
> Sleep should be implemented in HEAD, which commit did you build?

Yes, sorry. I was messing with an old commit.

Now is working properly with latest commit.

Thanks

closed 2020-12-05 by joraco-dev