kbeckmann/game-and-watch-retro-go #10: Pause when power off or on

macohime, 2020-12-06:
Like oridginal firmware's games.
I think when before power off or power on, behave just pushed key start(game) can provide this function.

nezza, 2020-12-06:
Which commit are you running? This is already implemented

macohime, 2020-12-06:
Latest. Save state works fine.I'm saying game is paused is better when game resumes.

kbeckmann, 2020-12-06:
Ah I think I got it. So when you press the power button it should simulate a press of the Start button? So when it wakes up the game will be paused. 

Could add this as an optional feature. For some software this is probably not always a good idea.

nezza, 2020-12-06:
Yea for a lot of games this will break: I.e. Start also *starts* a game in a lot of games

macohime, 2020-12-06:
>>kbeckmann
You think is right!!

>>nezza
Ahh... Yes. Start also start games... mmm... Perhaps we can't detect there is a title screen or playing game... mmm...

macohime, 2020-12-06:
How about keep start button was pressed or not since emulation has started?

uschmann, 2020-12-17:
What about pausing emulation until a Button is pressed after resuming?

fnordsh, 2021-01-09:
Right now it seems that powering off overwrites my savegame which is slightly annoying.
Could we have an extra save slot for "current state of whatever game was running when going to sleep" that is restored on power on, and then simulate a press on Pause/Set so we are in a kind of paused state, where we left off, and still can use the normal save slot independently?

kbeckmann, 2021-01-09:
> Could we have an extra save slot

That means we use 2x more space for saves, which is probably not ideal for people with many roms.

Another solution is to keep the state in ram and resume in a smarter way. RAM is retained when in deep sleep but it requires some work to be able to resume properly.

fnordsh, 2021-01-09:
> That means we use 2x more space for saves
You don't need to save *every* game on suspend, right? So that would be only one more slot, plus the information which game was running. A smarter resume sounds like the more elegant solution, but I'm unable to say if it is worth the extra work or not.

kbeckmann, 2021-01-09:
Good point. However this reusable suspend slot needs to be large enough to fit the largest save state which is 192kB iirc.

macohime, 2021-01-09:
I wanna play last game just one click the power button.

To skip rom manager and resume the state, I think this is the easiest way to imprement.
1. keep the rom title when the rom is booted.
2. If returned to rom manager, discard the title.
3. If the title was kept when rom manager boot up by pressing power button, boot it with state save, else just boot rom manager.

I think this is more likely behave of original G&W.

commit 4be40715 (2021-07-01): emulators: Render frames and pause on power resume

When the user presses the power button from the off
state, the last played game will load. A few frames
will be rendered and the emulator will be paused.

Closes #10
closed 2021-07-01 by kbeckmann