kbeckmann/game-and-watch-retro-go #28: Config state is not preserved

ntrpnr, 2020-12-27:
**Describe the bug**
Prereq: Have at least one ROM available on the G&W

Whenever the GB or NES emulator is exited using "Quit" or reloaded, the following config values are reset to their default value:
* Brightness
* Volume
* Probably 'Palette' as well, haven't tried it.
* Semi-related:
    * It would also be nice if the G&W remembers the last emulator that was used and exits to this emulator page (NES / GB)
    * It would also be extra nice if the G&W remembers which ROM that was played and puts the cursor at this ROM whenever a ROM is exited.

**Expected behavior**
The config values listed should be a global setting that is preserved until manually modified.

kbeckmann, 2020-12-27:
Thanks for the details. This is on the TODO :)

trochr, 2021-01-23:
I remember you live coding this, but I didn't  find it in the YT archived streams.  Would you kindly push the progress there in a branch from which I could attempt to take over ?
I think fixing this would bring a great refinement to the project

kbeckmann, 2021-01-23:
@trochr Thanks! Yeah sure please go ahead. Pushed my WIP branch now, feel free to use it as inspiration. https://github.com/kbeckmann/game-and-watch-retro-go/tree/kb/resume

kbeckmann, 2021-01-23:
Uh, sorry but it won't build. Seems I lost my changes in the submodule. Think I mostly added appid-defines. But again, please use the above mentioned branch as inspiration. I think what i got stuck at was a nice way to resume into the correct emulator if the user had pressed the power button while playing a game.

trochr, 2021-01-25:
I was able to get your branch working on the old version, at least the volume and brightness settings persitent.
 
Here is my fork which contains the changes added to the current main:
https://github.com/trochr/game-and-watch-retro-go/commit/117714a30b2353356c23711c048dcd87fa14877a

I tried to place the configflash sections at different places, it either ignored (config is not perstisted) or it goes to BSOD when launching a Rom :
https://github.com/trochr/game-and-watch-retro-go/blob/117714a30b2353356c23711c048dcd87fa14877a/STM32H7B0VBTx_FLASH.ld#L190

The BSOD shows a magic mismatch on this line:
https://github.com/trochr/game-and-watch-retro-go/blob/117714a30b2353356c23711c048dcd87fa14877a/Core/Src/porting/odroid_settings.c#L87

If you could give your insight, I would appreciate the help !

closed 2021-02-07 by kbeckmann