kbeckmann/game-and-watch-retro-go #35: Gameboy Color Roms seem to be locked to palette 2/9 and don't display correctly

deanhuff, 2020-12-28:
I am having an issue with [GB] when running [Tetris DX (JU) [C][!].gb].
Git hash of main repo [heads/main-0-gd34667c-dirty] and submodule [tags/1.20.2-32-g234052c].
My problem is: 

It seems that Gameboy color roms always use pallete 2/9 and don't display correctly.
Additionally, it would be nice if the python script parse_roms.py supported file extension gbc.

deanhuff, 2020-12-29:
[image]

deanhuff, 2020-12-29:
[image]

Correct color example.

kbeckmann, 2020-12-29:
Great with a correct reference. Something is seriously messed up in the GBC colors :). Will have a look at it tomorrow.

kbeckmann, 2020-12-29:
This was a great example btw, as there are a lot of colors to use as a reference.

kbeckmann, 2020-12-29:
Solved in 29451a18c12b53d7855c3107b5c4de23d9de7b8a

closed 2020-12-29 by kbeckmann