kbeckmann/game-and-watch-retro-go #43: Issues with similar ROM files.

DNA64, 2021-01-04:
```heads/main-0-g52c6150-dirty``` 
```remotes/origin/nintendogw-main2-0-g635d7421```

### ROM 1
ROM: `Rad Racer 2 (USA).nes` 
CRC32: `801836A4` 
Issue: BSOD  
Size: 64k PRG / 64k CHR 
Mirroring: `H` 
Mapper: `MMC3 (iNES 004)`

**Describe the bug**
At first glance the game appears to work. However, after pressing start, and selecting a stage, as soon as you press the `Game` button you'll see the BSOD. This can actually result in a system that doesn't respond to button pushes, meaning you have to disconnect the battery. Though thankfully, this level of a crash only happened to me once and I wasn't able to reproduce it, so it should be rare. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start new game
2. Press start `Game`
3. Press start again .
4. See error

**Expected behavior**
Not a BSOD.

### ROM 2
ROM: `Family Tennis (Japan).nes`
CRC32: `E80B8D07`
Issue: Major Graphical Issues
Size: 64k PRG / 64k CHR
Mirroring: `V`
Mapper: `MMC3 (iNES 004)`

**Describe the bug**
Mapper/Sprite Errors?

**To Reproduce**
Steps to reproduce the behavior:
1. Start new game
2. Wait for demo to start (Or press start and play)
3. See error

**Expected behavior**
Full Tennis Court view, no garbage on screen.

### ROM 3
ROM: `Family Boxing (Japan).nes` 
CRC32: `02140A08`
Issue: Major Graphical Issues
Size: 64k PRG / 64k CHR
Mirroring: `V`
Mapper: `MMC3 (iNES 004)`

**Describe the bug**
Mapper/Sprite Errors? Black Screen/Missing Sprites.

**To Reproduce**
Steps to reproduce the behavior:
1. Start new game.
2. See the error

**Expected behavior**
Graphics are usually nice to have ;)

kbeckmann, 2021-01-04:
Investigated ROM: Rad Racer 2 (USA).nes. The problem is that the emulator has a bug where it is *writing* to the ROM. I don't know enough about how the emulator works to properly fix it (but could work around it)

kbeckmann, 2021-01-04:
The emulator doesn't handle Family Tennis (Japan).nes well. Can reproduce in Linux, not a porting issue.