kbeckmann/game-and-watch-retro-go #52: Tetris GB Crash, invalid opcode

Koitere, 2021-01-06:
I am having an issue with [gb] when running [Tetris_(JUE)_(V1.1)_[!].gb].
Git hash of main repo [heads/main-0-gedef752] and submodule [heads/nintendogw-main2-0-g14b1154].
My problem is: Game causes immediate crash, citing invalid opcode (picture included). Hash is sha256sum.
https://cdn.discordapp.com/attachments/612851795580157992/796488006189449246/image0.jpg

Desktop: macOS Catalina 10.15.4
Debugger: ST-Link v2 on Nucleo-64 board

How to reproduce: Launch game from gnuboy inside retro-go which should cause immediate crash.

Expected behaviour: Game launches successfully

kbeckmann, 2021-01-08:
Is this still valid?

From discord:
> Weird, Tetris started working and is no longer giving the invalid op code error despite nothing being changed

> despite nothing being changed
Did you just reboot the device and it started working, or did you flash it again?

Koitere, 2021-01-09:
Was just a device reboot, no re flashing needed. Will close the issue

closed 2021-01-09 by Koitere