kbeckmann/game-and-watch-retro-go #59: 'A' button is pressed in-game upon resume

ntrpnr, 2021-01-10:
Repro steps:

1. Open a ROM
2. Let the ROM be in a state where the 'A' button will trigger something (e.g. a jump in SMB)
3. Press Pause/Set
4. Select Save & Quit
5. Open the same ROM again using 'Resume Game'

Result: The 'A' Button is triggered (e.g. Mario jumps in SMB). This can be an unwanted behavior in many cases (e.g. the game character dies directly upon loading the saved state).

Expected: The 'A' button being triggered in-game should be suppressed upon resume

jan2642, 2021-01-14:
@ntrpnr Could you verify that this issue is now fixed and close it ? Thanks!

ntrpnr, 2021-01-16:
@jan2642 Great, will have a look soon

kbeckmann, 2021-01-21:
This works great now, thanks for the help implementing it in #62 .

closed 2021-01-21 by kbeckmann