kbeckmann/game-and-watch-retro-go #63: Sounds for Gameboy not working properly

djbriane, 2021-01-14:
I am having an issue with [GB] when running [tobu.gb].
Git hash of main repo [heads/main-0-g1670080] and submodule [remotes/origin/nintendogw-main2-0-g982dfa1].

When flashing Gameboy roms (I've tried several and they all have issues) the sound does not work correctly. On the same device with NES the sound works perfectly well. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash any GB rom to the device.
2. Start the game and compare the audio to an actual Gameboy (Tetris is one that is immediately obvious).

**Expected behavior**
Sounds would be the same as from the original Gameboy, instead they are nearly unrecognizable from their original.
 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Debugger: ST-Link v2 (original)

**Additional context**
Prior to around December 26th the sound on Gameboy was working perfectly well so something changed between then and around January 9th (I was on holiday so wasn't keeping up with builds).

kbeckmann, 2021-01-14:
Can you please record a video or provide exact filename + hashes of the ROMs that give you problems? I can't reproduce. Here's a capture of `Tetris (W) (V1.1) [!].gb` sha256sum `0d6535aef23969c7e5af2b077acaddb4a445b3d0df7bf34c8acef07b51b015c3`. Sounds good to me but maybe it should sound differently?

https://user-images.githubusercontent.com/3134352/104590194-3ffd8200-566b-11eb-9fe3-8de01a3c683f.mp4

djbriane, 2021-01-14:
Seems my version of Tetris is the same as yours: `0d6535aef23969c7e5af2b077acaddb4a445b3d0df7bf34c8acef07b51b015c3  Tetris (W) (V1.1) [!].gb`

This is a video showing the difference between the old build (12/26) and latest build:
https://streamable.com/4ljv9q

kbeckmann, 2021-01-14:
Weird, I don't have any explanation to this. Could you try to make a clean build and load only the Tetris rom and nothing else? i.e. `make clean`, and only have the tetris rom in `roms/gb`, then build and flash again.

Also, which gcc version are you using? 10.2.x is known to be good. Here's mine:

```
$ arm-none-eabi-gcc --version                      
arm-none-eabi-gcc (Arch Repository) 10.2.0
```

djbriane, 2021-01-14:
Same result when I flash just Tetris by itself. 

gcc version (I updated this recently in hopes it'd fix the issue):
`arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)`

Output of compilation here: https://pastebin.com/qv9wbfc0

Here is my flash output just in case it makes any difference:
```
[12:05:35] 2 $ make LARGE_FLASH=1 flash
[ BASH ] Checking for updated roms
Entering 'retro-go-stm32'
../game-and-watch-flashloader/flash_multi.sh build/gw_retro_go_extflash.bin
0
else
Preparing chunk 0 in file /tmp/flash_chunk.zOzR8S
124+0 records in
124+0 records out
126976 bytes transferred in 0.000837 secs (151687822 bytes/sec)
Flashing!
Loading image into RAM...
Open On-Chip Debugger 0.10.0+dev-01517-g7b641d3d4-dirty (2020-12-03-14:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.856992
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Resetting device
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800bf04 msp: 0x20020000
Programming ELF
13860 bytes written at address 0x00000000
1736 bytes written at address 0x20000000
downloaded 15596 bytes in 0.159698s (95.370 KiB/s)

Loading image into RAM
126976 bytes written at address 0x24000000
downloaded 126976 bytes in 1.312112s (94.504 KiB/s)

sp (/32): 0x20020000

pc (/32): 0x00003561

Starting flash process
Please wait til the screen blinks once per second.
(Rapid blinking means an error occured)
Done!
[ BASH ] Checking for updated roms
Entering 'retro-go-stm32'
[ BIN ] gw_retro_go_intflash.bin
openocd -f ../game-and-watch-flashloader/interface_stlink.cfg -c "program build/gw_retro_go_intflash.bin 0x08000000 verify reset exit"
Open On-Chip Debugger 0.10.0+dev-01517-g7b641d3d4-dirty (2020-12-03-14:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.860158
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800bf04 msp: 0x20020000
** Programming Started **
Info : Device: STM32H7Ax/7Bx
Info : flash size probed value 128
Info : STM32H7 flash has a single bank
Info : Bank (0) size is 128 kb, base address is 0x08000000
Warn : Adding extra erase range, 0x0800f1e0 .. 0x0800ffff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

djbriane, 2021-01-14:
Few things I've tried in the meantime (with the same results):
1. Flash without LARGE_FLASH flag
2. Flash with ST-Link v2 clone (I had one lying around)
3. Install `arm-none-eabi-gcc` 10.2.1 from the MacOS package on the ARM site (I had used homebrew previously)

djbriane, 2021-01-15:
Still debugging but so far I've figured out that something changed between:

12465740367d4be18807ed551332572921349148 (gb audio works fine) 

and 

debeffe721294afbda92ca415583c657577fc323 (gb audio is broken)

Going to start digging into the changes here and see if I can find exactly what caused it. #

djbriane, 2021-01-15:
Well I reverted the changes from the retro-go-stm32 project dealing with memory allocation but now am getting this:

```
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: build/gw_retro_go.elf section `._ram' will not fit in region `RAM'
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 159760 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [build/gw_retro_go.elf] Error 1
make: *** [flash_all] Error 2
```

@kbeckmann any thoughts on what might be going on here? I am a bit out of my depth in this code but it seems like this change is the one most likely causing the issue (or at least blocking me from digging in further): https://github.com/kbeckmann/retro-go-stm32/commit/14b1154d4618a359712c58147c91368150e32b99

kbeckmann, 2021-01-15:
The change you are referring to was made together with a change in this repo that relocated the emulators to 0x24000000 and then loading the appropriate code+data from flash when starting the emulator.

You could comment out the lines in Core/Src/retro-go/rg_emulators.c:emulator_start() where the nes and sms emulators are loaded. That way the other emulators will not be linked in and you should not get a ram overflow.

djbriane, 2021-01-18:
I tried commenting out those lines but still ran into the ram overflow. Upon review, changes in the 0b553b87da91900c6a708924ec492fb2080e646d commit are pretty extensive so it may be beyond me to work out what exactly is causing the audio issues. I may try flashing via Linux VM to see if its a macOS issue (only reason I could think that no one else is reporting this as an issue).

kbeckmann, 2021-01-18:
Could you make a build with the latest main branch, and only this rom https://demozoo.org/productions/171568/ placed in `roms/gb` and then upload the resulting `build/gw_retro_go.elf`, `build/gw_retro_go_intflash.bin` and `build/gw_retro_go_extflash.bin` ? Then I can take a look and try to figure out what's different from when I build it.

djbriane, 2021-01-19:
Here you go! 
[gw_retro_go.zip](https://github.com/kbeckmann/game-and-watch-retro-go/files/5833031/gw_retro_go.zip)

kbeckmann, 2021-01-19:
Thanks!

I loaded your binaries to my device and it played fine, no audio issues that I can hear. But I build the same code with `xpack-arm-none-eabi-gcc-10.2.1-1.1-linux-x64` which should be equivalent to your compiler, however I get a different result. Used ghidra to decompile all the functions and diffed our code, and from what I can see they were mostly located in different areas.

I'll attach my build here. And a video of _your_ build running on my device. As I said earlier, I can't hear a problem, but maybe I'm missing something?

[retro-go-kbeckmann.zip](https://github.com/kbeckmann/game-and-watch-retro-go/files/5833741/retro-go-kbeckmann.zip)

https://user-images.githubusercontent.com/3134352/104990625-4306c280-5a1d-11eb-8411-9d6a32bd3ae3.mp4

Also just as a double check - make sure you select "New Game" and not "Resume game" as you might have some old corrupt save data.

To flash my build onto your device you may run the following commands:

`../game-and-watch-flashloader/flash_multi.sh gw_retro_go_extflash.bin`

`openocd -f interface/stlink.cfg -c "transport select hla_swd;" -f target/stm32h7x.cfg -c "adapter speed 500; init; program gw_retro_go_intflash.bin 0x08000000 verify reset"`

djbriane, 2021-01-19:
Just flashed your build onto my device and it is having the same audio problems - here is a video (it is quite obvious comparing your video to mine):

https://streamable.com/y2qn2n

No idea what this means - one might think its a hardware problem but as I said I have no problems with audio for NES roms (and if I revert to 1246574 the GB audio is fine).

kbeckmann, 2021-01-20:
This is really the weirdest issue I've seen the last few months. I'm not sure what to do about this. It seems there is some hardware difference between our devices, but it still doesn't explain why the issue disappears when you run an older build.

It seems one of the audio channels just isn't being played for you, for some reason. I pulled in a few commits from upstream, maybe you can try checking out and building the following commit: bc61c642b13d8fe35e6d8bf592d1cf25f625bdae . Make sure to run `git submodule update --init` after you checked it out.

djbriane, 2021-01-21:
Checked out the commit but I'm unable to get it to load the rom properly (have tried flashing it multiple times with the same result). 

[image]

My console output during flash:
```
[20:23:04] 2 $ make -j4 LARGE_FLASH=1 flash
[ BASH ] Checking for updated roms
Entering 'retro-go-stm32'
openocd -f ../game-and-watch-flashloader/interface_stlink.cfg -c "program build/gw_retro_go.elf reset exit"
Open On-Chip Debugger 0.10.0+dev-01517-g7b641d3d4-dirty (2020-12-03-14:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.878812
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800bee8 msp: 0x20020000
** Programming Started **
Info : Device: STM32H7Ax/7Bx
Info : flash size probed value 128
Info : STM32H7 flash has a single bank
Info : Bank (0) size is 128 kb, base address is 0x08000000
Info : Padding image section 1 at 0x080100d8 with 8 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x080100e0 .. 0x08011fff
Warn : no flash bank found for address 0x20000e0c
Warn : no flash bank found for address 0x24000000
Warn : no flash bank found for address 0x2404b000
Warn : no flash bank found for address 0x30000000
Warn : no flash bank found for address 0x90000000
Warn : no flash bank found for address 0x90000278
Warn : no flash bank found for address 0x9003a010
Warn : no flash bank found for address 0x900497f8
Warn : no flash bank found for address 0x90ff3000
** Programming Finished **
** Resetting Target **
shutdown command invoked
```

djbriane, 2021-01-21:
Same results with that commit - I'll probably go back and try using Docker or a VM to flash via Linux as thats the only thing I can think of to try (long shot I realize).

djbriane, 2021-01-22:
Running via Docker (according to your instructions) worked and GB sound is now working as expected! So I guess its some kind of issue between macOS and Ubuntu?

kbeckmann, 2021-01-22:
> Running via Docker (according to your instructions) worked and GB sound is now working as expected! So I guess its some kind of issue between macOS and Ubuntu?

Good to hear, but I don't understand why you weren't able to get a good result when using my image. That should've worked. This sounds like there is some issue with either flashloader or openocd on mac. In flashloader, there are two bash scripts that do the heavy lifting. There is a risk that some of the calculations become incorrect on mac for some reason. 

The scripts are https://github.com/ghidraninja/game-and-watch-flashloader/blob/main/flash_multi.sh and https://github.com/ghidraninja/game-and-watch-flashloader/blob/main/flash.sh if you want to have a look.

djbriane, 2021-01-22:
This makes sense as I noticed when I compiled locally, then ran the docker build command (which copies all of the build files over to the container) and flashed from docker the audio issue wasn't present. I'll poke around in the flashloader and see what I can find out.

kbeckmann, 2021-01-22:
Interesting observation! Yeah, seems like the script is indeed broken on macOS then. Ideally maybe they should be written in python so we don't run in to bash-flavor issues. Doing "fancy" math in bash feels wrong anyway, and I have a gut feeling that's where the issue is.

jan2642, 2021-01-24:
I'm also using macOS but never had this issue. On the other hand I'm not using flashloader, I flash directly with openocd. I'll double-check by using flashloader. I'm really curious if I can reproduce this issue.

kbeckmann, 2021-01-25:
@jan2642 

> I flash directly with openocd

If you feel up for it, it would you like to make a patch that gets rid of flashloader and uses openocd instead? Flashloader itself is sort of a pain to maintain and has been the source of many problems as you know :).

djbriane, 2021-01-26:
Just to confirm, I flashed from macOS using openOCD and the sound issue is not present.

Unfortunately I am not able to flash builds made with LARGE_FLASH=1 via openOCD but thats a separate issue (happy for some tips @jan2642)

jan2642, 2021-01-26:
The size of the flash is configured in the OCTOSPI peripheral of the SoC. If you use a larger flash this needs to be configured.

Replace this line:
`mww 0x52005008 0x01140208				;# OCTOSPI_DCR1: MTYP=0x1, DEVSIZE=0x14, CSHT=0x2, DLYBYP=0x1`
with this:
`mww 0x52005008 0x01180208				;# OCTOSPI_DCR1: MTYP=0x1, DEVSIZE=0x18, CSHT=0x2, DLYBYP=0x1`

DEVSIZE indicates the size of the flash:
    `size = pow(2, DEVSIZE + 1)`
According to openocd docs there's a silicon bug in the SoC where weirdness could happen near the end of the region. Because of this they advise to set DEVSIZE one larger than the actual size. I this case:
    `pow(2, 0x18 + 1) = pow(2, 25) = 33.554.432 = 16.777.216 * 2`

djbriane, 2021-01-26:
Thanks! However, after flashing it gives me a BSOD whenever I launch a game. Could this be due to an incorrect DEVSIZE value (I am using the standard Macronix 128Mb flash chip)?

Here is the output I get at the terminal:

```
Save data:	1253376 bytes
ROM data:	13148800 bytes
Total:		14402176 / 16777216 bytes (plus some metadata).

[10:56:52] $ openocd -f interface/stlink.cfg -c "transport select hla_swd" -f gw.cfg -c "init" -c "reset_config none" -c "reset halt" -c "flash write_image erase build/gw_retro_go.elf" -c "flash verify_image build/gw_retro_go.elf" -c "reset halt" -c "resume" -c "exit"
Open On-Chip Debugger 0.10.0+dev-01517-g7b641d3d4-dirty (2020-12-03-14:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.135050
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
none separate

target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800bf74 msp: 0x20020000
Info : Device: STM32H7Ax/7Bx
Info : flash size probed value 128
Info : STM32H7 flash has a single bank
Info : Bank (0) size is 128 kb, base address is 0x08000000
Initializing Octo-SPI interface
Info : valid SFDP detected
Info : flash1 'sfdp' id = 0x3825c2 size = 16384kbytes
Info : Padding image section 1 at 0x08010b0c with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08010b10 .. 0x08011fff
Warn : Adding extra erase range, 0x90cb6c08 .. 0x90cbffff
auto erase enabled
wrote 13399832 bytes from file build/gw_retro_go.elf in 493.299164s (26.527 KiB/s)

Info : Padding image section 1 at 0x08010b0c with 4 bytes (bank write end alignment)
Error: verify failed in bank at 0x90000000 starting at 0x00000000
```

jan2642, 2021-01-26:
OpenOCD complained:
`Error: verify failed in bank at 0x90000000 starting at 0x00000000`
That's probably why it crashes when starting games. Maybe if you lower the adapter speed a bit ?
The verify step should also succeed.

djbriane, 2021-01-27:
How can I lower the adapter speed via OpenOCD (I can do this in flashloader but don't see anything obvious in the cfg or openocd docs)?

djbriane, 2021-01-28:
Lowering adapter speed solved it and the Gameboy sound issue is now gone!

closed 2021-02-07 by kbeckmann