kbeckmann/game-and-watch-retro-go #75: Recovered NES Simcity glitching and showing wrong tiles

JuliusS123, 2021-01-27:
Emulator: NES
ROM name: simcity.nes
Git hash of main repo [heads/main-0-g8b5dd9a] and submodule [heads/nintendogw-main2-0-g982dfa1].

Bug description:
When the simcity ROM is launched the emulator gliches out, showing wrong tiles and colors. There are no issues with the music

Steps to reproduce the behavior:
Simply flash the simcity rom to the G&W. Upon load the above listed things occour.

Expected behavior:
I have run the game in "higan" (another nes emulator) and it worked as expected.
There is a screenshot of it working here:
[image]
This is what it looks like on my G&W:
[image]
[image]

My OS is debian-based(linux mint)
I am using the cinnamon desktop(default for the system)

I have tried reflashing the rom and nothing has changed. I can also play other games perfectly fine.

ChrisHughes, 2021-02-02:
I'm seeing similar glitching on other MMC5 games like Castlevania 3.

rzinurov, 2021-03-15:
I saw a similar behavior when I "resumed" the newly added ROM after I re-flashed the GW with a new set of ROMs. The loaded state of the new ROM contained graphics and parts of the screen of a different ROM, which state I saved before I re-flashed. In my case it wasn't ROM-specific and it looked like a corrupted state of the saved data. Try deleting the save from the menu and then saving/loading the game. Chances are - it will work just fine, in which case the real issue is related to persisting the save states when re-flashing the device.