kbeckmann/game-and-watch-retro-go #80: Feature request: auto power off

rzinurov, 2021-03-15:
Hi there! First of all I'd like to thank you for such an amazing project. I've been using it for week now and I'm absolutely loving it. Really impressive work!

Now the feature request. Yesterday when I tried to power on my GW the screen stayed dark. It powered on with the charger and showed empty battery indicator, despite the fact that the battery was fully charged when I used it last time. I suspect that I simply forgot to power it off and it drained the battery over night. How difficult would it be to add auto power off feature when there is no activity (no buttons pressed?) for let's say 10 minutes while in game or after 1 minute in the menu?

kbeckmann, 2021-03-15:
Yeah it sounds likely that this is what happened. I’ve been thinking of this too and guess it should be a configurable setting. Some people probably want it to automatically save before going to standby, and others may not want their save state to be overwritten. PRs are welcome :)

rzinurov, 2021-03-16:
I'd love to give it a shot, but I have no idea where to even start. I have a solid Java/Python/JS experience, but I've never done any microcontroller programming before. What would be the best way to get started? Ideally I'd need an IDE with syntax highlight/completion and some sort of a device emulator.

kbeckmann, 2021-03-16:
I personally use VSCode and develop mostly for the target directly. The iteration cycle is reasonably fast (a few seconds). Just make your changes and run 
`make -j$(nproc) install`. To make it even faster, keep only one small rom in your roms directory.

To debug with GDB you can run `make openocd` first (to start an openocd instance and attach to the debugger), and then in another terminal run `make gdb` which starts gdb and talks to the device through the openocd instance.

rzinurov, 2021-03-16:
All right, sounds like I could give it a try. I guess it's time to get myself a proper debugger, because so far I've been using Raspberry Pi. Can you by any chance recommend a good (and budget) debugger model, that would work with mac os and is available in Sweden (or can be shipped here from within the EU)?

commit c73bd95d (2021-07-02): Suspend after idling in the main menu

Default timeout is 10 minutes.

Also adds a configurable timeout for this.

Closes #80
closed 2021-07-02 by kbeckmann