kbeckmann/game-and-watch-retro-go #88: Request for additional language file name support

apple524, 2021-05-05:
At present, the name of the game directory can only be in English. I hope to add Japanese and Chinese support。
Thank you for your contribution to this project！

Arnold101knight, 2021-05-09:
ITALIAN TOO. i can translate

kbeckmann, 2021-05-09:
> At present, the name of the game directory can only be in English

Do you mean the name of the ROMs can only be in English? If so, then there are multiple issues with this.

- objdump seems to have issues with Japanese symbol names, they would have to be mangled/escaped. Example: Calling a rom "マリオ.nes" results in an object name `_binary__home_konrad_dev_game_and_watch_retro_go_roms_nes___________nes_lz4_start` so that's not great.
- There is no font to render non-English characters. See https://github.com/kbeckmann/retro-go-stm32/blob/nintendogw-main2/components/odroid/bitmaps/font_basic.h

> I hope to add Japanese and Chinese support

Cool, PRs are welcome. But I am slightly worried about how much space the font will take up. Maybe rendering the game name to static images in the build process would be a way to keep things small.

> ITALIAN TOO. i can translate

@Arnold101knight Please open a separate issue to track localization, as in translating the UI to various languages. This is not something I am planning on working on myself though, but feel free to contribute.

olderzeus, 2021-09-10:
> ITALIAN TOO. i can translate

https://github.com/olderzeus/game-and-watch-retro-go/tree/i18n

You can try to add ITALIAN. maybe need a font file if current font file doesn't support.