kbeckmann/game-and-watch-retro-go #90: make -j4 issue with gcc-10 and gcc-9 installed (Ubuntu 20.2.0)

rataweb, 2021-06-14:
Hi,
With Ubuntu 20.2.0 if I have two versions of gcc (gcc-9 and gcc-10), "make -j4" doesn't work. It always shows the following message:

> "Please use GCC Version 10
> 
> You are using:
> gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1)

GCC-10 Installed and configured as default:

> $ sudo update-alternatives --config gcc
>
> There are 2 choices for the alternative gcc (providing /usr/bin/gcc).
>
>Selection    Path            Priority   Status
>   0            /usr/bin/gcc-10   100        auto mode
>  **1            /usr/bin/gcc-10   100        manual mode**
>   2            /usr/bin/gcc-9   90        manual mode

And GCC version:

> $ gcc --version
>gcc (Ubuntu 10.2.0-5ubuntu1~20.04) 10.2.0
>Copyright (C) 2020 Free Software Foundation, Inc.
>This is free software; see the source for copying conditions.  There is NO
>warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Thanks and Best regards,
Shark

nezza, 2021-06-14:
What's the version of your arm gcc?

```
arm-none-eabi-gcc --version
```

Thanks and cheers,
Thomas

kbeckmann, 2021-06-14:
Ubuntu doesn't package gcc 10 for arm afaik, so I recommend downloading arm-gcc-none-eabi 10 from over here https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads .

You can then extract is somewhere and set the environment variable `GCC_PATH` to the bin directory, e.g. `/opt/gcc-arm-none-eabi-10-2020-q4-major/bin`.

rataweb, 2021-06-14:
Hi,
You're right I had arm-none-eabi-gcc 9.2.1 version. Downloading and configuring new version now it works!!!

Thanks a lot for your help,
Shark

kbeckmann, 2021-06-14:
Great!

closed 2021-06-14 by kbeckmann