kbeckmann/game-and-watch-retro-go #97: GG (Sega Game Gear) doesn't work

olderzeus, 2021-07-03:
All gg rom run an error occurred

FATAL EXCEPTION: Hardfault 5244218-dirty
PC=0x***    LR=0x**

kbeckmann, 2021-07-03:
What are your local changes? The `-dirty` tells us that there are changes.

Make sure that you've run `git pull --recurse-submodules` to update the submodules.

When you see the problem again, take a photo of the screen and post it please as it contains helpful information.

olderzeus, 2021-07-04:
I pull a new version agin， same error.
rom download from romhustler.org.

see the Img: [https://imgtu.com/i/RWifhR](url)

kbeckmann, 2021-07-04:
I just tested a few GG roms together with NES and GB (saw that you have that setup) and it works for me. Can you post a callstack from your system?

First make sure to run `make flash`, then in one console, run `make openocd`. In another, run `make gdb` and then type `bt` in the console and post the result.

Should look like this: https://asciinema.org/a/rgIy0jcXkbaHUhzX4K8L22uf4

olderzeus, 2021-07-04:
(gdb) bt
#0  0x08000730 in buttons_get () at Core/Src/gw_buttons.c:11
#1  0x08001324 in BSOD (fault=65, fault@entry=BSOD_HARDFAULT, pc=0x100da2f, lr=0x800d796 <memset+10>) at Core/Src/main.c:178
#2  0x080041a2 in HardFault_Handler () at Core/Src/stm32h7xx_it.c:96
#3  <signal handler called>
#4  0x0800d796 in memset ()
#5  0x08005422 in emulator_start (file=<optimized out>, load_state=<optimized out>, start_paused=<optimized out>) at Core/Src/retro-go/rg_emulators.c:404
#6  0x08005638 in emulator_show_file_menu (file=0x80119a8 <gg_roms>) at Core/Src/retro-go/rg_emulators.c:354
#7  0x080056a2 in event_handler (event=<optimized out>, tab=0x2000ce7c) at Core/Src/retro-go/rg_emulators.c:75
#8  0x080050ba in retro_loop () at Core/Src/retro-go/rg_main.c:337
#9  0x08005178 in app_main () at Core/Src/retro-go/rg_main.c:385
#10 0x08002044 in main () at Core/Src/main.c:534

kbeckmann, 2021-07-04:
Ah thank you. I can reproduce it now by having a *lot* of roms.

olderzeus, 2021-07-04:
Thank you for the all

kbeckmann, 2021-07-04:
I am looking into this, but the fix will just be a better validation of the memory usage, i.e. it will complain and tell you that you are using too many roms. If you look at the build output you probably see something like this

```
ram_emu_nes	103408 / 671744	(568336 bytes free (0.542 MB))
ram_emu_gb	242348 / 671744	(429396 bytes free (0.410 MB))
ram_emu_sms	673172 / 671744	(-1428 bytes free (-0.001 MB))
ram_emu_pce	0 / 671744	(671744 bytes free (0.641 MB))
```

note the negative value. I will add proper validation now, but in the longer term we will try to use less ram for the rom information.

kbeckmann, 2021-07-04:
So what you can do right now is to remove roms until you don't see any negative size values.

kbeckmann, 2021-07-04:
Alright, I have now fully understood the issue and added asserts in 52187019472cf91efdeb47a30dbfd7610529f8ff. It is not related to number of roms.

On my system where I use GCC 11.1.0, I reproduced it by enabling debug information which increased the RAM utilization. However, it seems that the current code will not fit when built with GCC 10.2.0. I'll see if I can squeeze it down a bit.

kbeckmann, 2021-07-04:
It's fixed now. Please run `git pull --recurse-submodules` and try again.

commit 4fc0a3c4 (2021-07-04): Update submodule

coleco: Store BIOS in extflash only (no copy to RAM)

Fixes #97
closed 2021-07-04 by kbeckmann