kbeckmann/game-and-watch-retro-go #116: Use the RTC for timekeeping in emulators

kbeckmann, 2021-07-26:
This would be for gb in particular.

Will probably also need some GUI to set the date/time.

northskysl, 2021-08-14:
GUI to set the RTC time was implemented in the STM32 RTC patchset that was merged to 'main'. Getting the RTC functionality to work proper with GB emulator would likely require SRAM saving to be implemented first. The RTC patchset has an exported function that can be used to return a 64-bit Unix timestamp, which could be useful when implementing RTC support to the GB emulator. Here is useful documentation about the RTC save format in GB: https://bgb.bircd.org/rtcsave.html

kbeckmann, 2021-08-14:
Do you want to look into the SRAM saving? If not, I could take a stab at it.

northskysl, 2021-08-14:
@kbeckmann I do not have the needed time currently, but it would be great if you could take a look! This would be a nice feature to have