kbeckmann/game-and-watch-retro-go #117: Improve audio when running emulator at 0.5x

kbeckmann, 2021-07-29:
When running at 0.5x the sound could be better.

One major issue is that we end up only updating one of the double buffers with new data, which means that we get stuck with 50% old samples which doesn't sound good. We could copy the data into the alternate buffer when swapping buffers (since some emulators use the audio buffer as a synch mechanism) every second swap or something like that. Seems easy on the surface but could get a bit messy.

kbeckmann, 2021-07-29:
@BrianPugh I noticed this while testing #114. No rush but wanted to let you know.

deanhuff, 2021-07-29:
.5x speed??  I'll be able to cheat my way to NES Tetris Level 9 5 rows.  :)

BrianPugh, 2021-07-29:
@deanhuff it's currently available, give the latest builds a try, but yah as kbeckmann said the audio could be better; I'll see what I can do.