kbeckmann/game-and-watch-retro-go #137: GCC 10 not found on RPi 3

rablador, 2021-08-18:
This is not really a bug, more of a setup problem. I've tried REAL hard to have GCC 10 be detected while compiling the project on a Raspberry Pi 3, but I never get past the compiler check.

More specifically:

```
Please use GCC Version 10

You are using:
-e
```

I've tried downloading and setting GCC_PATH manually (to no avail), as described here: https://github.com/kbeckmann/game-and-watch-retro-go/issues/90

kbeckmann, 2021-08-18:
That error message means that make can't execute `$GCC_PATH/arm-none-eabi-gcc`.

Make sure that you have arm-none-eabi-gcc downloaded and extracted somewhere. Try to find exactly where the binary `arm-none-eabi-gcc` is on your system and point GCC_PATH to that directory.

rablador, 2021-08-18:
Yep, that's what I've been trying:

1. [Locating arm-none-eabi-gcc](https://www.dropbox.com/s/c2ybsede4az7goh/Screenshot%202021-08-18%20at%2014.10.11.png?dl=0)
2. [Setting GCC_PATH and running make](https://www.dropbox.com/s/n42wyaimvvse3qe/Screenshot%202021-08-18%20at%2014.11.21.png?dl=0)

kbeckmann, 2021-08-18:
That’s weird. I pushed a change that should print a better error message. Also can you run `make help` and post the output?

rablador, 2021-08-18:
Pulled latest, updated all subrepos, ran make clean. Still same result (but with function ref a few lines down: `make: *** [Makefile.common:376: CheckTools] Error 1`). Here's the make help output:

```
Game and Watch Retro Go

Configuration variables:
  EXTFLASH_FORCE_SPI  - Forces the use of legacy SPI mode for the external flash driver
  EXTFLASH_SIZE_MB    - Sets the external flash size in MB
  EXTFLASH_SIZE       - Sets the external flash size in bytes (deprecated)
  LARGE_FLASH         - Sets the external flash size to 16MB  (deprecated)
  EXTFLASH_OFFSET     - Places the data at an offset in the external flash (useful for dual boot)
  INTFLASH_BANK       - Sets the internal flash bank. Valid values {1,2} (default=1).
  RESET_DBGMCU        - Configures if DBGMCU should be reset after flashing.
                        Set to 0 to disable power saving (default=1)

Current configuration:
  EXTFLASH_FORCE_SPI=0
  EXTFLASH_SIZE_MB=1
  EXTFLASH_SIZE=1048576
  LARGE_FLASH=0
  EXTFLASH_OFFSET=0
  INTFLASH_BANK=1
  RESET_DBGMCU=1
  GCC_PATH=/opt/gcc10/gcc-arm-none-eabi-10.3-2021.07/bin
  PREFIX=arm-none-eabi-

Targets:
  flash_extflash    - Only programs the external flash
  flash_intflash    - Only programs the internal flash
  flash_intflash_nc - Only programs the internal flash and uses an existing openocd server
  flash_test        - Runs a flash test. Will overwrite data on the external flash!
  gdb               - Starts gdb and attaches to openocd
  gdb_intflash      - Runs flash_intflash_nc, then starts gdb and attaches to openocd
  openocd           - Starts openocd with appropriate config
  reset_dbgmcu      - Resets the unit and turns off DBGMCU (lowers battery drain)
  reset_mcu         - Resets the unit
  dump_logs         - Dumps the callstack and logbuf. Starts openocd and gdb under the hood.
```

rablador, 2021-08-18:
Hmm, is this relevant?

`-bash: /opt/gcc10/gcc-arm-none-eabi-10.3-2021.07/bin/arm-none-eabi-gcc: cannot execute binary file: Exec format error`

kbeckmann, 2021-08-18:
Yes, very much so. It seems that toolchain is not meant for your architecture.

kbeckmann, 2021-08-18:
You need a toolchain that is built for your architecture (the raspberry pi 3), that targets arm-none-eabi / armv7e-m.

rablador, 2021-08-18:
Ah, that makes sense! Should have checked that first, but I blindly just grabbed the one referenced in ticket #90... -_-
Will try to get one for rpi then. Thanks for your help!

rablador, 2021-08-18:
Alright, got one from xpm (`xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest`) that works.

kbeckmann, 2021-08-18:
Cool. In case xpack isn't installed and you want to grab the binaries directly you can get it here https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/

Same with openocd https://github.com/xpack-dev-tools/openocd-xpack/releases/

closed 2021-08-18 by kbeckmann