kbeckmann/game-and-watch-retro-go #139: can't make

chen010311, 2021-08-23:
HI~
OS:MACOS 11.5
DEBUGGER :ST-LINK
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.07) 10.3.1 20210621 (release)

game-and-watch-retro-go wumengjia$ make -j$(sysctl -n hw.logicalcpu) flash
[ WGET ] stm32h7xx.h
[ WGET ] stm32h7b0xx.h
[ WGET ] system_stm32h7xx.h
[ WGET ] cmsis_compiler.h
make[1]: *** [Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7b0xx.h] Error 3
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [Drivers/CMSIS/Device/ST/STM32H7xx/Include/system_stm32h7xx.h] Error 3
make[1]: *** [Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7xx.h] Error 3
make[1]: *** [Drivers/CMSIS/Include/cmsis_compiler.h] Error 3
make: *** [flash] Error 2
game-and-watch-retro-go wumengjia$ make clean
[ RM ] build
game-and-watch-retro-go wumengjia$ make LARGE_FLASH=1 flash
[ WGET ] stm32h7b0xx.h
make[1]: *** [Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7b0xx.h] Error 3
make: *** [flash] Error 2

kbeckmann, 2021-08-23:
Can you check if you have `wget` installed? It’s used to download the drivers.

chen010311, 2021-08-23:
game-and-watch-retro-go wumengjia$ brew install wget
Updating Homebrew...
Warning: wget 1.21.1_1 is already installed and up-to-date.
To reinstall 1.21.1_1, run:
  brew reinstall wget

kbeckmann, 2021-08-23:
Okay. 

Can you try running `make download_sdk` first?

If that doesn’t work either could you run `make VERBOSE=1` and post the full output?

chen010311, 2021-08-23:
KUAIRUIdeAir:game-and-watch-retro-go wumengjia$ make download_sdk
[ WGET ] stm32h7xx_hal_adc.c
[ WGET ] stm32h7xx_hal_adc_ex.c
[ WGET ] stm32h7xx_hal_cortex.c
[ WGET ] stm32h7xx_hal_dac_ex.c
[ WGET ] stm32h7xx_hal_dac.c
[ WGET ] stm32h7xx_hal_dma_ex.c
[ WGET ] stm32h7xx_hal_dma.c
[ WGET ] stm32h7xx_hal_exti.c
[ WGET ] stm32h7xx_hal_flash_ex.c
[ WGET ] stm32h7xx_hal_flash.c
[ WGET ] stm32h7xx_hal_gpio.c
[ WGET ] stm32h7xx_hal_hsem.c
[ WGET ] stm32h7xx_hal_i2c_ex.c
[ WGET ] stm32h7xx_hal_i2c.c
[ WGET ] stm32h7xx_hal_ltdc_ex.c
[ WGET ] stm32h7xx_hal_ltdc.c
[ WGET ] stm32h7xx_hal_mdma.c
[ WGET ] stm32h7xx_hal_ospi.c
[ WGET ] stm32h7xx_hal_pwr_ex.c
[ WGET ] stm32h7xx_hal_pwr.c
[ WGET ] stm32h7xx_hal_rcc_ex.c
[ WGET ] stm32h7xx_hal_rcc.c
[ WGET ] stm32h7xx_hal_rtc_ex.c
[ WGET ] stm32h7xx_hal_rtc.c
[ WGET ] stm32h7xx_hal_sai_ex.c
[ WGET ] stm32h7xx_hal_sai.c
[ WGET ] stm32h7xx_hal_spi_ex.c
[ WGET ] stm32h7xx_hal_spi.c
[ WGET ] stm32h7xx_hal_tim_ex.c
[ WGET ] stm32h7xx_hal_tim.c
[ WGET ] stm32h7xx_hal_wwdg.c
[ WGET ] stm32h7xx_hal.c
[ WGET ] startup_stm32h7b0xx.s
KUAIRUIdeAir:game-and-watch-retro-go wumengjia$ make VERBOSE=1
/usr/bin/env bash -c " \
		for tool in /Applications/ARM/bin/arm-none-eabi-gcc /Applications/ARM/bin/arm-none-eabi-objcopy /Applications/ARM/bin/arm-none-eabi-size /Applications/ARM/bin/arm-none-eabi-gdb /Applications/ARM/bin/arm-none-eabi-objdump; do \
		which \$tool 2> /dev/null > /dev/null || \
		( \
			echo -e \"\nCannot find \$tool\n\" && \
			exit 1 \
		) \
		done"
/Applications/ARM/bin/arm-none-eabi-gcc -v 2>&1 | grep "gcc version 1[01]" > /dev/null || \
		( \
			echo -e "\n\n\nPlease use GCC version 10\n\n\nYou are using:" && \
			/Applications/ARM/bin/arm-none-eabi-gcc -v 2>&1 | grep "gcc version" ; \
			echo -e "\n\n\n" && \
			exit 1 \
		)
/Applications/ARM/bin/arm-none-eabi-objcopy --info | grep "elf32-littlearm" > /dev/null || \
		( \
			echo -e "\n\n\n/Applications/ARM/bin/arm-none-eabi-objcopy does not suport elf32-littlearm.\nPlease install a toolchain that supports elf32-littlearm and armv7e-m.\n\n\n" && \
			exit 1 \
		)
/Applications/ARM/bin/arm-none-eabi-objdump --help | grep "armv7e-m" > /dev/null || \
		( \
			echo -e "\n\n\nobjdump does not suport armv7e-m.\nPlease install a toolchain that supports elf32-littlearm and armv7e-m.\n\n\n" && \
			exit 1 \
		)
echo [ BASH ] Checking for updated roms
[ BASH ] Checking for updated roms
./scripts/update_rom_files.sh build/rom_files.txt roms/gb roms/nes roms/sms roms/gg roms/col roms/pce roms/sg roms/gw
echo [ PYTHON3 ] rom_files.txt
[ PYTHON3 ] rom_files.txt
/usr/bin/env python3 parse_roms.py --flash-size 1048576 --compress=lz4
Compressing: Nintendo Gameboy / 1:   0%|                  | 0/2 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/Users/wumengjia/game-and-watch-retro-go/parse_roms.py", line 720, in <module>
    ROMParser().parse(args)
  File "/Users/wumengjia/game-and-watch-retro-go/parse_roms.py", line 559, in parse
    save_size, rom_size = self.generate_system(
  File "/Users/wumengjia/game-and-watch-retro-go/parse_roms.py", line 491, in generate_system
    self._compress_rom(
  File "/Users/wumengjia/game-and-watch-retro-go/parse_roms.py", line 451, in _compress_rom
    output_file.write_bytes(output_data)
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/pathlib.py", line 1265, in write_bytes
    with self.open(mode='wb') as f:
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/pathlib.py", line 1242, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/pathlib.py", line 1110, in _opener
    return self._accessor.open(self, flags, mode)
PermissionError: [Errno 13] Permission denied: '/Users/wumengjia/game-and-watch-retro-go/roms/gb/1.gb.lz4'
make: *** [build/roms.a] Error 1

kbeckmann, 2021-08-23:
Permission denied indicates that you might have run `sudo make` earlier. Either remove the files, change permissions or run as sudo to go forward.

chen010311, 2021-08-23:
Thanks a lot！You solved my problem.

closed 2021-08-23 by chen010311

kbeckmann, 2021-08-23:
Great!