kbeckmann/game-and-watch-retro-go #155: Does the Retro-Go use the same resolution as native game?

laoshanxi, 2021-10-17:
Please include the following:

- Which console :NES
- The full name of the ROM you are running, Tank, almost all the ROMs

**Describe the bug**
Compared with the original game, the resolution is 320*240 that the font is clear in the game, but I found in the Retro-Go emu, the WIDTH is less than 320 which cause the font is not much clear, see the screenshots I uploaded

**To Reproduce**
I have no debug platform, I am using a machine which already make everything ready with 100+ NES games by someone with 64M Storage, and the system is already CN language, so I want to known whether this repo have the same resolution compared with native game. 

**Expected behavior**
The game Resolution is the same as native game for NES.

**Screenshots**
[image]

kbeckmann, 2021-10-17:
The display is 320x240. For NES, we scale it up to 307x240 because it looks pretty good that way. You can change scaling algorithm here https://github.com/kbeckmann/game-and-watch-retro-go/blob/main/Core/Src/porting/nes/main_nes.c#L25

closed 2021-10-17 by kbeckmann

laoshanxi, 2021-10-18:
> 
> 
> I added support for changing the scaling and filtering algorithms from the menu in [b51f0f9](https://github.com/kbeckmann/game-and-watch-retro-go/commit/b51f0f99d5cd0e8911afb120cab35fe8e08052d3).
> 
> The available algorithms for GB are: Off/Fit : no stretching, borders on the sides. Full - filtering Off/Sharp: sharp bilinear-ish stretching, no borders. The previous and default algorithm. Full - filtering Soft: soft bilinear stretching, no borders. Custom: aka "JTH", compressed top and bottom sections, super sharp middle section.
> 
> The available algorithms for NES are: Off/Fit: No stretching, borders on the sides. Full - filtering Off: No borders, Nearest-neighbor stretching, no borders. Full - filtering Sharp: No borders, Bilinear-ish sharp stretching. Custom: Almost fills the screen, sharp. The previous and default algorithm.

Wow, this is pretty good. thanks in advance!