kbeckmann/game-and-watch-retro-go #163: Screenshot feature ?

myStphane, 2021-10-29:
Could it be possible to have a screenshot feature to capture the current screen with a combination of buttons?

Idea: reserve a memory space (in advance, like save space) to have "some" slot(s) for screenshot(s), and then later retrieve it (& extract it as jpg / png / raw image) like the script "saves_backup.sh" does.

kbeckmann, 2021-10-29:
Yes, shouldn't be too hard. Easiest would be to have a static allocation and do a raw frame buffer copy, as the size will be predictable and fixed. JPEG is lossy, and PNG could potentially become large based on the pixel data.

bzhxx, 2021-10-30:
we can also respawn the framebuffer capture over stlink to get some capture.

kbeckmann, 2021-10-30:
You can stream the framebuffer in real time to a virtual camera in Linux today, and use it in e.g. obs. It’s not fast but works :). https://github.com/kbeckmann/game-and-watch-retro-go/blob/main/tools/screengrabber.py

olderzeus, 2021-10-30:
use gdb and run :
`dump binary memory img1.bin 0x24000000 0x24025800`
`dump binary memory img2.bin 0x24025800 0x2404B000`
will get raw bmp file data but not file header.

Dump that file and finally run: 
`copy img0.bin/b+img1.bin/b img.bmp `
at windows will (img0.bin is a bmp file (320x240 RGB565)'s header) get the screen bmp image file.

kbeckmann, 2021-10-31:
@myStphane I have now implemented the feature. 

Screenshots can be captured by pressing `PAUSE/SET` + `GAME`. This feature is disabled by default if the external flash is 1MB (stock units), because it takes up 150kB in the external flash.

Screenshots can be downloaded by running `make dump_screenshot`, and will be saved as a 24-bit RGB PNG.

commit ee3cb5e9 (2021-10-31): Add screenshot support

Closes #163
closed 2021-10-31 by kbeckmann