kbeckmann/game-and-watch-retro-go #181: Compression for GG roms

retrohead, 2022-01-05:
Could you add compression for GameGear roms please if possible?

bzhxx, 2022-01-05:
I have implemented it for sega emulators using either ITCRAM or flash reserved space. Retrohead, if you want you can try it using my repo. If it's convenient, I can push it.

FrackME, 2022-08-02:
> I have implemented it for sega emulators using either ITCRAM or flash reserved space. Retrohead, if you want you can try it using my repo. If it's convenient, I can push it.

hey @bzhxx, i'm  new to the GnW stuff, how can i use the compression for sega emulators with your repo? i already use it but no compression after all 

keep up the good work guys, you are awesome