kbeckmann/game-and-watch-retro-go #189: Major slowdowns on The Guardian Legend [NES]

SantX27, 2022-03-20:
Please don't attach or link to game ROMs and similar files unless the license explicitly allows this.

Please include the following:

- Which console (NES, GB, ...)
**NES**

- The full name of the ROM you are running, e.g. "Super_Tilt_Bro_(E).nes"
**Guardian Legend, The (USA).nes, Guardian Legend, The (Europe).nes**

- The git hash of this repo and the submodule. Run the following: `git describe --all --long --dirty=-dirty; cd retro-go-stm32; git describe --all --long --dirty=-dirty`
**Can't do this for now, hope it's not required. I have problems even with olderzeus's fork as well**

**Describe the bug**
**When a lot of enemies are on screen (which on this game happens most of the time), the performance tanks a lot, until they go away. This does not happen on real hardware (proof: https://www.youtube.com/watch?v=ImwZBSo8Rvo)**

**To Reproduce**
**Simply boot up the game and start a new game, in less than a minute, when the blue meteorites start appearing and then chasing you, the framerate drops tremendously, while the audio is virtually unchanged.**

**Expected behavior**
**It should not slow down, like the real NES**

**Screenshots**
**I'm not really sure how to export a performance graph, so I can't help here**

**Desktop environment:**
- Host device OS: **Artix Linux**
- Debugger: **ST-Link v2 clone**

**Additional context**
Clearly a nofrendo problem, probably with the mappers (or with the rendering)?

SantX27, 2022-03-24:
I have tried debugging this issue on my own, using GDB and it turns out that I can't use it, I don't know how to debug a running program. I've seen in the nofrendo code some profiling lines and some printfs, but these don't show up on GDB, maybe I have to enable them at compilation/runtime? I have also tried compiling the 2.0pre1 of nofrendo on my PC, however it refuses to run, so maybe I don't have luck with this specific emulator 😄. Any direction on how to solve this problem is very appreciated, I really want to get this game running right and I know for a fact that the GnW has the guts to do it.
In the meantime I've finished the game, but all the lag in the final stages is very annoying, so I really hope I can get it running properly

kbeckmann, 2022-04-07:
Sorry for the late response.

If you would like to debug this on a desktop machine in Linux, you can do the following:

```bash
cd linux
./update_nes_rom.sh ~/roms/nes/Guardian\ Legend,\ The\ \(E\)\ \[\!\].nes
make -f Makefile.nes -j
./build/retro-go-nes.elf

# If you want to debug with gdb:
gdb ./build/retro-go-nes.elf
(gdb) break <function name>
(gdb) run

```

The integration does not handle sound properly, and there could be a lot of other issues with it too. It was made to ease porting in the beginning of the project. But, it works for debugging issues on desktop.

If you want to debug on hardware, it's possible. However, it's a bit more complicated because we support multiple emulators. To ease debugging, make sure to only have one rom in the roms folder. We started to write a [developer guide in the wiki](https://github.com/kbeckmann/game-and-watch-retro-go/wiki/Developer-Guide#debug-on-target), but it's very bare bones for now.

Also if you haven't already, feel free to join the discord and discuss in #emulators .

SantX27, 2022-04-17:
Ok, I've done a bit of digging around in the source code, and I found out that the main culprit is `apu_emulate()` in `nes_emulate()` inside `nes.c`. Disabling it, other than obviously muting all sounds of the game, fixes every stuttering I had, but (and I don't even know how or why) makes the game's status bar (pictured and circled in red below) flicker when there are a lot of enemies on screen (Trust me, it doesn't happen on real hardware). Honestly, I'm not competent enough to dive inside the APU code trying to figure out what causes both problems. And while it is a "solution", you might agree with me that it is kinda stupid fixing the game by disabling the APU like this 😆 . I'm starting to think that this problem might even afflict the upstream Retro-GO, but I don't have an Odroid to test the ROM with, so I'm not going to make an upstream issue. Also thanks for the discord invite, but I don't think it's a good platform to make bug reports with, since they always get lost. Thank you again!
[image]