kbeckmann/game-and-watch-retro-go #190: Having problems.

lowskill32, 2022-03-26:
Lowskill@ubuntu:~/game-and-watch-retro-go$ make
Cannot find arm-none-eabi-gdb
-e 

Please use GCC version 10
You are using:
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1) 
-e 
make: *** [Makefile.common:414: CheckTools] Error 1

How can i fix this issue. Trying for my self, but no success.

SantX27, 2022-03-26:
It seems that you haven't properly configured the toolchain, but that depends on some factors, nominally Ubuntu version and platform used. Are you compiling on a Raspberry? If this is the case, I think it can be compiled but there isn't a guide for it, yet.

lowskill32, 2022-03-26:
I'm using stlink. This is my second day to try find solution.

SantX27, 2022-03-26:
I've got that, but that is the flasher, I'm more interested in knowing on which machine you are running make on, so that I can guide you. Is it a Raspberry or another SBC? Is it a desktop PC?

lowskill32, 2022-03-26:
Runing on desktop pc using virtula machine.

mplinuxgeek, 2022-03-26:
@lowskill32 github issues are for reporting bugs with software you should take this question to a more appropriate forum, your initital error shows you are using a gcc version that's older than required so its not a bug, you need to download a newer version of gcc.

You can get more direct support in the discord or reddit:
https://discord.com/channels/781528730304249886/783282561001717771
https://www.reddit.com/r/GameAndWatchMods/

lowskill32, 2022-03-27:
Solved my problems and now everything works great.

kbeckmann, 2022-03-27:
Glad you got it to work, and thanks for helping out @mplinuxgeek

closed 2022-03-27 by kbeckmann