kbeckmann/game-and-watch-retro-go #192: Restoring saves dow

jzobac, 2022-05-18:
Hi,
when trying to restore saves (seems that backup works) i get only "timeouts".

configuration
rpi 3, Zelda handheld, 64MB flash from Farnell (4MB original patched fw, 60MB retrogo) 

Raspberry pi OS
2022-04-04-raspios-bullseye-armhf-lite.img.xz (32bit)
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

compile command of retrogo
export OPENOCD="/opt/openocd-git/bin/openocd"
export GCC_PATH=/home/pi/opt/xpack-arm-none-eabi-gcc-10.3.1-2.3/bin
export ADAPTER=rpi
export adapter=rpi

time make -j$(nproc) GNW_TARGET=zelda EXTFLASH_SIZE_MB=60 EXTFLASH_OFFSET=4194304 INTFLASH_BANK=2 ENABLE_SCREENSHOT=0 flash 2>&1 | tee log.txt
      

git describe --all --long --dirty=-dirty
heads/main-0-g1925eb6

retro-go-stm32; git describe --all --long --dirty=-dirty
heads/main-0-g5f1dd2b

Here is log from restoring saves
Programming save for:
    rom_name="Alien 3"
    save_address=67076096
    save_size=28672

[0;32mPreparing chunk 1 / 1 in file /tmp/flash_chunk.uxWgr8[0m
[0;32mFlashing![0m
Open On-Chip Debugger 0.11.0+dev-00685-g830d70bfc-dirty (2022-05-15-14:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'sysfsgpio swd_nums' not 'sysfsgpio_swd_nums'
DEPRECATED! use 'sysfsgpio srst_num' not 'sysfsgpio_srst_num'
none separate

Info : SysfsGPIO JTAG/SWD bitbang driver
Info : This adapter doesn't support configurable speed
Info : SWD DPIDR 0x6ba02477
Info : [stm32h7x.cpu0] Cortex-M7 r1p1 processor detected
Info : [stm32h7x.cpu0] target has 8 breakpoints, 4 watchpoints
Info : gdb port disabled
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0801b64c msp: 0x2001b620
0x2001b620
0x0801b64d
msp (/32): 0x2001b620

pc (/32): 0x0801b64d

Starting flash app
State: FLASHAPP_INIT
//
'handheld powers off
//
State: UNKNOWN
State: UNKNOWN
State: UNKNOWN
State: UNKNOWN
State: UNKNOWN
State: UNKNOWN
//
'I power him on, from original fw switch to retro-go
//
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
State: FLASHAPP_INIT
//
'that's all, nothing restores, just init loop

With best regards
jzobac

jzobac, 2022-05-18:
restore command
export ADAPTER=rpi
export OPENOCD="/opt/openocd-git/bin/openocd"
export GCC_PATH=/home/pi/opt/xpack-arm-none-eabi-gcc-10.3.1-2.3/bin
export adapter=rpi

time ./scripts/saves_restore.sh build/gw_retro_go.elf 2>&1 | tee log-restore.txt

ashimokawa, 2022-06-04:
Same problem here - with the original 4MB Zelda flash chip.

kbeckmann, 2022-06-06:
Make sure to set the correct environment variables when you restore the save games, just like you do when you run `make flash`.

It's likely you have not set e.g. `INTFLASH_BANK=2`.

I just did a full test of a clean Zelda unit with 64MB flash and an ST-Link, and it works.

My commands:

```
export EXTFLASH_SIZE_MB=60
export ADAPTER=stlink
export GNW_TARGET=zelda
export INTFLASH_BANK=2

# From game-and-watch-patch:
make PATCH_PARAMS="--device=zelda" flash_patched

# From game-and-watch-retro-go:
make -j8 EXTFLASH_SIZE_MB=60 EXTFLASH_OFFSET=4194304 INTFLASH_BANK=2 flash

# Play some games, then backup saves
make flash_saves_backup

# Save some new state to know that you have a different save locally vs on the device

# Restore locally stored saves
make flash_saves_restore

# Verify on the device

```

closed 2022-06-06 by kbeckmann

jzobac, 2022-06-14:
Yes, works'
Thank you.