kbeckmann/game-and-watch-retro-go #195: Extract save file / compatibility of state save with other emulators

cryptoluks, 2022-09-20:
Thanks for the great work, it works flawlessly for my GBC roms.

I wonder if there any feasible (hacky) possibility, to use the state saves or part of it in any other emulator at a later point. Would it be possible to parse the state saves to build a generic sav file?

When doing an in-game (SRAM save?), is this data actually preserved in the GW retro go state save backup and therefore possible extractable if knowing the position/structure?

Thanks for any hint or suggestion.

kbeckmann, 2022-09-20:
You can see the details of the save file here:
https://github.com/kbeckmann/retro-go-stm32/blob/main/gnuboy-go/components/gnuboy/loader.c#L831

You could ask in the discord if someone has already written a converter as
I know other people have asked for the same thing.

cryptoluks, 2022-09-23:
The battery save seems to be simply appended starting at offset `0xD000`, as also indicated by the gnuboy saved state file header.

So extraction with, for example, dd: `dd if=./saved-state.save of=./battery-save.sav bs=1 skip=53248`

Import to a saved state should also be possible, by using `seek=53248` on the saved state output file.

closed 2022-09-23 by cryptoluks